gazebosim/gazebo_test_cases #1306: gz-sim: Create an underwater vehicle

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/underwater_vehicles.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/underwater_vehicles.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

knmcguire, 2024-09-13:
Working on this test case with windows 11

knmcguire, 2024-09-13:
✅ Works fine!

as before, I had to change the texture location on both the .dae as the .sdf file as both relative and .. paths didn't work on windows. And since there were so many wgets, I used `curl.exe` this time to retrieve all the files! That worked like a charm.

[image]

ahcorde, 2024-09-13:
Thank you for testing!

closed 2024-09-13 by ahcorde