gazebosim/gazebo_test_cases #1308: gz-sim: Using Components in a System Plugin

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/usingcomponents.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/using_components.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-02:
Interested

Combuster54, 2024-09-04:
## Ubuntu 24.04 Noble

**Case Study: Using the JointForceCmd Component**

[image]

Combuster54, 2024-09-04:
**Case Study: Using the Pose Component**

There are no suggestions.

Everything seems to be fine

closed 2024-09-04 by iche033