gazebosim/gazebo_test_cases #1312: gz-sim: Video Recorder

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/videorecorder.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/video_recorder.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-16:
Everything works as it should (except OGV recording - ffmpeg says it doesn't have vp8 encoder).

azeey, 2024-09-16:
It seems like the conda version of ffmpeg doesn't enable vp8. It explicitly does `--enable-libvpx` for MacOS and Linux (see https://github.com/conda-forge/ffmpeg-feedstock/blob/main/recipe/build.sh). I think a note in the tutorial would be sufficient.

peci1, 2024-09-19:
I've added the note here: https://github.com/gazebosim/gz-sim/pull/2634 .

peci1, 2024-09-19:
The behavior when OGV is selected on Windows is suboptimal, though. In the best case, the format should be greyed out or not offered at all.

Currently, I can select the format, and then the GUI console gets spammed:

```
(2024-09-20 01:36:14.413) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.429) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.430) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.445) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.446) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.461) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.462) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.478) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.479) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.494) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.495) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.511) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.512) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.528) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.529) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
(2024-09-20 01:36:14.546) [info] [VideoRecorder.cc:224] [GUI] Recording video using bitrate: 2070000
(2024-09-20 01:36:14.547) [error] [VideoEncoder.cc:416] [GUI] Codec for[vp8] not found. Video encoding is not started.
```

traversaro, 2024-09-23:
> It seems like the conda version of ffmpeg doesn't enable vp8. It explicitly does `--enable-libvpx` for MacOS and Linux (see https://github.com/conda-forge/ffmpeg-feedstock/blob/main/recipe/build.sh). I think a note in the tutorial would be sufficient.

Just a cross-link to the issue(s) tracking libvpx (and hence vp8) support for Windows:
* https://github.com/conda-forge/ffmpeg-feedstock/issues/149
* https://github.com/conda-forge/libvpx-feedstock/pull/7

closed 2024-09-23 by ahcorde