gazebosim/gazebo_test_cases #1317: gz-gui: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/9/install.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/01_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-30:
## Ubuntu 24.04 (Noble)
### Binary Install
Everything works fine

### Source Install
Everything works fine

### Documentation
Everything works fine

### Testing

Sanjay1911, 2024-09-04:
[image]
I did the installation for testing other issues and test 15 failed for me. Maybe @trushant05 , can you check again?

j-rivero, 2024-09-04:
> The branch should reference to `-b gz-gui<#>` which is currently `-b ign-gui<#>` where <#> is the version. For this test I have used gz-gui8 branch.

could you please open a PR changing everything to gz- , please?

azeey, 2024-09-04:
> For this test I have used gz-gui8 branch.

Also, do you mind using the `gz-gui9` branch since that's what is in Ionic?

j-rivero, 2024-09-04:
> could you please open a PR changing everything to gz- , please?

Could you also please change the minimun required from Ubuntu Focal to Ubuntu Noble?

trushant05, 2024-09-10:
Hello @azeey, I just completed testing with ```gz-gui9``` branch and as @Sanjay1911 showed I got just one failed test as follow:

[image]

@j-rivero I have submitted the [PR](https://github.com/gazebosim/gz-gui/pull/639).

closed 2025-08-27 by azeey