gazebosim/gazebo_test_cases #1320: gz-gui: Command Line

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/9/commandline.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/02_commandline.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

csimon-be, 2024-08-30:
Interested in this one

Sanjay1911, 2024-09-04:
Not sure if @csimon-be tested this issue already, but every argument works as expected for me. This can be closed @j-rivero 

 

- [x]  -l [ --list ]           

- [x]   -s [ --standalone ] 

- [x]  -c [ --config ]

- [x]  -v [ --verbose ] 

- [x] -h [ --help ]      
                                            
- [x]   --force-version <VERSION>  Use a specific library version.

- [x] --versions                 Show the available versions.

azeey, 2024-09-04:
@csimon-be do you plan to test this?

csimon-be, 2024-09-05:
@azeey Indeed I was a bit delayed but I tested it, everything is working as expected for me either.

ahcorde, 2024-09-05:
Thank you for testing!

closed 2024-09-05 by ahcorde