gazebosim/gazebo_test_cases #1321: gz-gui: Command Line

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/gui/9/commandline.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/02_commandline.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

knmcguire, 2024-09-12:
I'm working on this now

knmcguire, 2024-09-12:
❌ Errors on windows 11

I get the following error message with all gz gui line commands except for the `--versions` and  `--force-versions`

```
<internal:C:/Users/.../miniconda3/envs/gz-ws/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require': cannot load such file -- C:/Users/kimbe/Development/simulation/gazebo/gz-ws/install/lib/ruby/gz/cmdgui9 (LoadError)
        from <internal:C:/Users/.../miniconda3/envs/gz-ws/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require'
        from C:/Users/kimbe/Development/simulation/gazebo/gz-ws/install/bin//gz:310:in `<main>'
```

Mind that `gz sim -g` does work

closed 2024-09-13 by azeey