gazebosim/gazebo_test_cases #1323: gz-gui: Plugins

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/9/plugins.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/03_plugins.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Mechazo11, 2024-09-06:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
     * In my humble opinion this tutorial may need to be presented in context of a more hands-on example. The current version was hard to follow, specially as a newcomer to Gazebo.
    * Add a note stating if built from source, ```GZ_GUI_PLUGIN_PATH``` can be set as follows
    ```bash
    cd ~/workspace/src/gz-gui
    export GZ_GUI_PLUGIN_PATH=${HOME}/workspace/src/gz-gui
    echo $GZ_GUI_PLUGIN_PATH
    gz gui -s ImageDisplay
    ```
    * Tutorial does not show how to set `GZ_GUI_PLUGIN_PATH`. Nor it states the user had to change direcotry into ```src/gz-gui``` directory before this command would work correctly
    

[image]

[image]

* **Issues**
    * Users needs to change directory to ```~/workspace/src/gz-gui```. Without this step, invoking any of example commands with ```gz gui -c``` prefix throws ```XMLError``` error.
    * The tutorial page has multiple ```\subpage``` references that leads to nowhere.
    * Commands for Topic Viewer, Topic Stats, Requesters and Responders did not work.

[image]

ahcorde, 2024-09-06:
https://github.com/gazebosim/gz-gui/pull/638

azeey, 2024-09-16:
Created https://github.com/gazebosim/gz-gui/issues/643 for improving the tutorial. Since the functionality works, I'll go ahead and close this.

closed 2024-09-16 by azeey