gazebosim/gazebo_test_cases #1329: gz-gui: Style

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/9/style.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/05_style.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alexandrosnic, 2024-09-03:
Interested in this issue

alexandrosnic, 2024-09-03:
At least for the docker container users of Gazebo, a default config file could not be found.
I would provide the users with an example of how to create one:
```
echo '<?xml version="1.0"?>

<window>
  <width>602</width>
  <height>767</height>
  <style
    material_theme="Dark"
    material_primary="Yellow"
    material_accent="#ff5555"
  />
</window>

<plugin filename="Publisher"/>

<plugin filename="WorldControl">
  <ignition-gui>
    <title>Controls: all</title>
  </ignition-gui>
  <play_pause>true</play_pause>
  <step>true</step>
  <start_paused>false</start_paused>
  <service>/world_control</service>
  <stats_topic>/world_stats</stats_topic>
</plugin>' > style.config 

gz gui -c style.config
```

and perhaps I would also link this [example config file](https://github.com/gazebosim/gz-gui/blob/ign-gui2/examples/config/style.config).

Other than that, looks good