gazebosim/gazebo_test_cases #1335: gz-gui: Config

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/9/config.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/07_config.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Mechazo11, 2024-09-03:
Hi @iche033 can you kindly unassign this test. Unfortunately don't have enough time to try this one out

Sanjay1911, 2024-09-04:
I can try this @azeey

Sanjay1911, 2024-09-04:
- [x] Config plugin
[image]

tested with time.config, works as expected..

This issue can be closed

closed 2024-09-04 by j-rivero