gazebosim/gazebo_test_cases #1341: gz-gui: 3D Scene

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/9/scene.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui9/tutorials/scene.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Mechazo11, 2024-09-04:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
    * ```scene_provider``` builds okay
    * ```scene3d``` does not run the following runtime error is seen
   
    * Steps to reproduce
   
    * In one terminal
    ```bash
    cd src/gz-gui/examples/standalone/scene_provider
    mkdir build && cd build
    cmake ..
    make -j4
    ./scene_provider
    ```
    * In another terminal
    ```bash
    source ~/workspace/install/setup.bash
    gz gui -c examples/config/scene3d.config
    ```

[image]

ahcorde, 2024-09-06:
When running 

```
gz gui -c examples/config/scene3d.config
```

you should be located in the gz-gui directory or fix the path to point the right config file

closed 2025-08-27 by azeey