gazebosim/gazebo_test_cases #1344: gz-launch: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/launch)
* [Source](https://github.com/gazebosim/gz-launch/blob/gz-launch8/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-01:
Interested in ! (let's review a last one for today)

avanmalleghem, 2024-09-01:
- manager -> manage
- installation tutorial (currently on 7, should be 8)
- There is no "testing" section
- Link https://gazebosim.org/libs/gazebo is dead (usage section)
- "Known issue of command line tools" can be removed ? or at least we can explain how the solved issue work.

Will do this really soon

avanmalleghem, 2024-09-02:
PR opened : https://github.com/gazebosim/gz-launch/pull/274

What about my last point ? ("Known issue of command line tools" can be removed ? or at least we can explain how the solved issue work.)

ahcorde, 2024-09-02:
> * "Known issue of command line tools" can be removed ? or at least we can explain how the solved issue work.

Any thoughts about this comment ? @azeey

azeey, 2024-09-03:
The sentence

> However, that environment variable only takes a single path, which means if the installations from source are in different locations, only one can be specified.

is no longer true since https://github.com/gazebosim/gz-tools/issues/8 has been fixed. But the overall limitation of using `gz` with a combination of packages from Debs and built from source is still there. The new workaround is to list all locations for `GZ_CONFIG_PATH`, so it would be great to update the tutorial.

avanmalleghem, 2024-09-04:
Ok, did it here : https://github.com/gazebosim/gz-launch/pull/275

closed 2024-09-04 by azeey