gazebosim/gazebo_test_cases #1350: gz-launch: Gazebo launch tutorial

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/launch/8/basics.html)
* [Source](https://github.com/gazebosim/gz-launch/tree/gz-launch8/tutorials/tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-01:
Interested in but is it normal that the "live" link is on version 8 ?

avanmalleghem, 2024-09-04:
- all links to examples are dead on live format
- typo "the the robot"
- can't start factory.gzlaunch (because missing gz-sim-state-publisher-system plugin)
- segfault on [sim_plugins.gzlaunch](https://github.com/gazebosim/gz-launch/blob/gz-launch8/examples/sim_plugins.gzlaunch)

Need to investigate a little bit more on these 2 issues. Will do this soon and create a PR then.

avanmalleghem, 2024-09-05:
- I opened a PR to fix links and typo : https://github.com/gazebosim/gz-launch/pull/276
- Starting from a clean docker container, the factory.gzlaunch works
- Still a segfault on sim_plugins.gzlaunch. I don't know how to tackle this one

iche033, 2024-09-06:
I created https://github.com/gazebosim/gz-launch/pull/277 to fix the crash.

ahcorde, 2024-09-06:
PR is merged, thank you for testing!

closed 2024-09-06 by ahcorde