gazebosim/gazebo_test_cases #1356: gz-math: Angle example

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/math/8/angle.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/angle.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

jc-sz, 2024-08-31:
interested

akky20, 2024-09-03:
## Works fine 
[image]

### If getting any error like this , installing c++ compiler using `sudo apt-get install g++` will fix the issue
[image]

closed 2024-09-03 by caguero