gazebosim/gazebo_test_cases #1359: gz-math: Color example

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/math/8/color.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/color.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

jc-sz, 2024-08-31:
interested

akky20, 2024-09-03:
## Run correctly NO error 

### An error occured but can be fix using the command `sudo apt-get install g++`
[image]

## here is the final output 
[image]