gazebosim/gazebo_test_cases #1361: gz-math: C++ Get Started

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/math/8/cppgetstarted.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/cppgetstarted.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-09-10:
## Mac OS

### C++ Get Started Guide

I have tested the step-by-step guide provided and everything works file.

[image]

I have one doubt though, to compile following code is give:
```
c++ $(pkg-config --cflags gz-math8) main.cpp -o main
```

Shouldn't it be:
```
g++ $(pkg-config --cflags gz-math8) main.cpp -o main

# OR

clang ++  $(pkg-config --cflags gz-math8) main.cpp -o main
```

azeey, 2024-09-13:
Hmm, not sure. Probably symlinked to clang++ like cc is symlinked to clang

closed 2024-09-13 by azeey