gazebosim/gazebo_test_cases #1369: gz-math: Triangle example

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/math/8/example_triangle.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/example_triangle.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Tiryoh, 2024-09-12:
It seems good!

[image]

azeey, 2024-09-12:
Thanks for testing!!

closed 2024-09-12 by azeey

Tiryoh, 2024-09-12:
Sorry I forgot to mention that I changed the procedure a bit, could you check it out as I created a PR?

https://github.com/gazebosim/gz-math/pull/632