gazebosim/gazebo_test_cases #1371: gz-math: Vector example

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/math/8/example_vector2.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/example_vector2.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

NickTziaros, 2024-08-31:
I can try this one

NickTziaros, 2024-09-01:
Running on Ubuntu Noble following the recommended docker instructions.

Got the following error after using `cmake ..`.

[image]

Fixed with `sudo apt-get install build-essential`.

###  **The Vector example runs as expected.**

[image]

## Suggestions

Might I suggest adding the following text or something similar 

> Be sure to replace <#> with a number value, such as 7 or 8, depending on which version you need.<#>.

under 

`sudo apt install ruby-gz-math<#>`

also adding the output of
 
`ruby vector2_example.rb` and  `ruby vector3_example.rb`

```
va = 1.000000 2.000000
vb = 3.000000 4.000000
vc = 3.000000 4.000000
vb += va: 4.000000 6.000000
vb.Normalize = 0.554700 0.832050
vb.Distance(va) = 1.249959
```
```
v =: 0.000000 0.000000 0.000000
v2 = 1.000000 0.000000 0.000000
v1 + v2 = 1.000000 0.000000 0.000000
v1.Distance(v2) = 1.000000
```

for clarity reasons.

caguero, 2024-09-03:
Thanks for the help! Could you create a PR including your suggestions?

ahcorde, 2024-09-04:
PR is merged, thank your for testing!

closed 2024-09-04 by ahcorde