gazebosim/gazebo_test_cases #1372: gz-math: Vector example

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/math/8/example_vector2.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/example_vector2.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Tiryoh, 2024-09-12:
It seems good!

[image]

ahcorde, 2024-09-12:
Thank you for testing!

closed 2024-09-12 by ahcorde

Tiryoh, 2024-09-12:
Sorry I forgot to mention that I changed the procedure a bit, could you check it out as I created a PR?

https://github.com/gazebosim/gz-math/pull/632