gazebosim/gazebo_test_cases #1375: gz-math: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/math/8/install.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-11:
## Windows 10

**Binary Instalation**

```sh
gz-ws) C:\Users\User\gazebo-ws\test\math>conda install libgz-math8 --channel conda-forge
Channels:
 - conda-forge
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - libgz-math8

Current channels:

  - https://conda.anaconda.org/conda-forge
  - defaults

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

(gz-ws) C:\Users\User\gazebo-ws\test\math>
```

there's no gz-math8 
```sh
(gz-ws) C:\Users\User\gazebo-ws\test\math>conda search libgz-math8 --channel conda-forge
Loading channels: done
No match found for: libgz-math8. Search: *libgz-math8*

CondaError: KeyboardInterrupt

^C¿Desea terminar el trabajo por lotes (S/N)? N

(gz-ws) C:\Users\User\gazebo-ws\test\math>conda search libgz-math* --channel conda-forge
Loading channels: done
# Name                       Version           Build  Channel
libgz-math7                    7.0.2      h63175ca_0  conda-forge
libgz-math7                    7.0.2      h63175ca_1  conda-forge
libgz-math7                    7.0.2      h63175ca_2  conda-forge
libgz-math7                    7.0.2      h63175ca_3  conda-forge
libgz-math7                    7.1.0      h63175ca_0  conda-forge
libgz-math7                    7.2.0      h63175ca_0  conda-forge
libgz-math7                    7.3.0      h63175ca_0  conda-forge
libgz-math7                    7.3.0      h63175ca_1  conda-forge
libgz-math7                    7.4.0      he0c23c2_0  conda-forge
libgz-math7                    7.5.0      he0c23c2_0  conda-forge
libgz-math7                    7.5.1      he0c23c2_0  conda-forge

```

**Source Installation**
```
(gz-ws) C:\Users\User\gazebo-ws\test\math>conda install eigen --channel conda-forge
Channels:
 - conda-forge
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: done

# All requested packages already installed.
[... 314 lines cut ...]
  Material.cc
  MecanumDriveOdometry.cc
  PID.cc
  Rand.cc
  RollingMean.cc
  RotationSpline.cc
  SemanticVersion.cc
  SignalStats.cc
  SpeedLimiter.cc
  SphericalCoordinates.cc
  Spline.cc
  Compiling...
  SplinePrivate.cc
  Stopwatch.cc
  Temperature.cc
  Vector3Stats.cc
  Angle.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line
   to improve linker performance
     Creating library C:/Users/User/gazebo-ws/test/math/gz-math/build/lib/Release/gz-math8.lib and object C:/Users/User/gazebo-
  ws/test/math/gz-math/build/lib/Release/gz-math8.exp
  Generating code
  Finished generating code
  gz-math8.vcxproj -> C:\Users\User\gazebo-ws\test\math\gz-math\build\bin\Release\gz-math8.dll
  Building Custom Rule C:/Users/User/gazebo-ws/test/math/gz-math/CMakeLists.txt
```

Combuster54, 2024-09-11:
**Tests**
```sh
(gz-ws) C:\Users\User\gazebo-ws\test\math\gz-math>ctest -R Ruby_TEST.rb
Test project C:/Users/User/gazebo-ws/test/math/gz-math
No tests were found!!!
(gz-ws) C:\Users\User\gazebo-ws\test\math\gz-math\test>ctest -R Ruby_TEST.rb
Test project C:/Users/User/gazebo-ws/test/math/gz-math/test
No tests were found!!!
```

azeey, 2024-09-11:
I have updated the gazebo_test_cases readme regarding Windows testing. You'll need to build all gazebo Ionic packages from source

Combuster54, 2024-09-12:
**Results**

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples\build>cmake ..
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Configuring done (0.4s)
-- Generating done (0.8s)
-- Build files have been written to: C:/Users/User/gazebo-ws/test/gz-math/examples/build

(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples\build>cd ..

(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples>cmake --build build --config Release
Versión de MSBuild 17.11.2+c078802d4 para .NET Framework

  additively_separable_scalar_field3_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\ad
  ditively_separable_scalar_field3_example.exe
  angle_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\angle_example.exe
  box_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\box_example.exe
  color_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\color_example.exe
  diff_drive_odometry.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\diff_drive_odometry.exe
  gauss_markov_process.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\gauss_markov_process.exe
  graph_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\graph_example.exe
  helpers_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\helpers_example.exe
  interval_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\interval_example.exe
  kmeans.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\kmeans.exe
  matrix3_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\matrix3_example.exe
  object.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\object.exe
  piecewise_scalar_field3_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\piecewise_sca
  lar_field3_example.exe
  polynomial3_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\polynomial3_example.exe
  pose3_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\pose3_example.exe
  quaternion_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\quaternion_example.exe
  quaternion_from_euler.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\quaternion_from_euler.e
  xe
  quaternion_to_euler.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\quaternion_to_euler.exe
  rand_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\rand_example.exe
  region3_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\region3_example.exe
  temperature_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\temperature_example.exe
  triangle_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\triangle_example.exe
  vector2_example.vcxproj -> C:\Users\User\gazebo-ws\test\gz-math\examples\build\Release\vector2_example.exe

(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples>
```

closed 2024-09-13 by azeey