gazebosim/gazebo_test_cases #1377: gz-math: Python Get Started

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/math/8/pythongetstarted.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/pythongetstarted.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alesof, 2024-08-31:
## Configuration
Running on Ubuntu 22.04 using the suggested docker setup. Gazebo version is 9.0.0~pre1.

## Process
Instructions works and output matches the description:
```
Distance from 1 3 5 to 2 4 6 is 1.7320508075688772
```

It could be made more modern/readable by using fstrings instead of format.

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde