gazebosim/gazebo_test_cases #1382: gz-math: Rotation example

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/math/8/rotation_example.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/rotation_example.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-31:
Interested

Combuster54, 2024-08-31:
## Configuration
Ubuntu 24.04 Noble

Compiling 
```bash
combuster@combuster54:~/ros2_ws/src$ git clone https://github.com/gazebosim/gz-math/ -b gz-math8
cd gz-math/examples
mkdir build
cd build
cmake ..
make
Cloning into 'gz-math'...
remote: Enumerating objects: 21146, done.
remote: Counting objects: 100% (1013/1013), done.
remote: Compressing objects: 100% (614/614), done.
remote: Total 21146 (delta 530), reused 776 (delta 396), pack-reused 20133 (from 1)
Receiving objects: 100% (21146/21146), 11.46 MiB | 2.78 MiB/s, done.
Resolving deltas: 100% (15292/15292), done.
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: /home/combuster/ros2_ws/src/gz-math/examples/build
[  2%] Building CXX object CMakeFiles/additively_separable_scalar_field3_example.dir/additively_separable_scalar_field3_example.cc.o
[  4%] Linking CXX executable additively_separable_scalar_field3_example
[  4%] Built target additively_separable_scalar_field3_example
[  6%] Building CXX object CMakeFiles/angle_example.dir/angle_example.cc.o
[  9%] Linking CXX executable angle_example
[  9%] Built target angle_example
[ 11%] Building CXX object CMakeFiles/box_example.dir/box_example.cc.o
[ 13%] Linking CXX executable box_example
[ 13%] Built target box_example
[ 15%] Building CXX object CMakeFiles/color_example.dir/color_example.cc.o
[ 18%] Linking CXX executable color_example
[ 18%] Built target color_example
[ 20%] Building CXX object CMakeFiles/diff_drive_odometry.dir/diff_drive_odometry.cc.o
[ 22%] Linking CXX executable diff_drive_odometry
[ 22%] Built target diff_drive_odometry
[ 25%] Building CXX object CMakeFiles/gauss_markov_process.dir/gauss_markov_process_example.cc.o
[ 27%] Linking CXX executable gauss_markov_process
[ 27%] Built target gauss_markov_process
[ 29%] Building CXX object CMakeFiles/graph_example.dir/graph_example.cc.o
[ 31%] Linking CXX executable graph_example
[ 31%] Built target graph_example
[ 34%] Building CXX object CMakeFiles/helpers_example.dir/helpers_example.cc.o
[ 36%] Linking CXX executable helpers_example
[ 36%] Built target helpers_example
[ 38%] Building CXX object CMakeFiles/interval_example.dir/interval_example.cc.o
[ 40%] Linking CXX executable interval_example
[ 40%] Built target interval_example
[ 43%] Building CXX object CMakeFiles/kmeans.dir/kmeans.cc.o
[ 45%] Linking CXX executable kmeans
[ 45%] Built target kmeans
[ 47%] Building CXX object CMakeFiles/matrix3_example.dir/matrix3_example.cc.o
[ 50%] Linking CXX executable matrix3_example
[ 50%] Built target matrix3_example
[... 107 lines cut ...]
  gz::math::Vector3d euler(q.Euler());
  printf(" W % .6f\n X % .6f\n Y % .6f\n Z % .6f\n", q.W(), q.X(), q.Y(), q.Z());
  std::cout << "\nto Rotation matrix\n";
  printf("   % .6f  % .6f  % .6f\n"
         "   % .6f  % .6f  % .6f\n"
         "   % .6f  % .6f  % .6f\n",
          m(0, 0), m(0, 1), m(0, 2),
          m(1, 0), m(1, 1), m(1, 2),
          m(2, 0), m(2, 1), m(2, 2));
}
```

**Output** 

```bash
 W  0.500000
 X  0.500000
 Y  0.500000
 Z  0.500000

to Rotation matrix
    0.000000  -0.000000   1.000000
    1.000000   0.000000  -0.000000
    0.000000   1.000000   0.000000
```

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde