gazebosim/gazebo_test_cases #1383: gz-math: Rotation example

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/math/8/rotation_example.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math8/tutorials/rotation_example.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

azeey, 2024-09-11:
I've updated the gazebo_test_cases readme regarding Windows testing. You'll need to build all the Gazebo Ionic packages from source.

Tiryoh, 2024-09-12:
I guess we need to update the instructions for Windows.

It worked with the following process

```
cd gz-math\examples
mkdir build
cd build
cmake ..
cd ..
cmake --build build --config Release
.\build\Release\quaternion_to_euler.exe 0.5 0.5 0.5 0.5
```

[image]

ahcorde, 2024-09-12:
PR https://github.com/gazebosim/gz-math/pull/631

Combuster54, 2024-09-12:
Thanks @Tiryoh!

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples>.\build\Release\quaternion_to_euler.exe 0.5 0.5 0.5 0.5
Normalizing Quaternion components:
  W 0.5
  X 0.5
  Y 0.5
  Z 0.5
to
  W 0.5
  X 0.5
  Y 0.5
  Z 0.5

Converting to Euler angles
 roll   1.570796 radians
 pitch -0.000000 radians
 yaw    1.570796 radians

 roll   90.000000 degrees
 pitch -0.000000 degrees
 yaw    90.000000 degrees

to Rotation matrix
    0.000000   0.000000   1.000000
    1.000000   0.000000   0.000000
    0.000000   1.000000   0.000000
```
```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples>.\build\Release\quaternion_from_euler.exe  0 0 1.57
Converting Euler angles:
 roll   0.000000 radians
 pitch  0.000000 radians
 yaw    1.570000 radians

 roll      0.000000 degrees
 pitch     0.000000 degrees
 yaw      89.954374 degrees

to Quaternion
 W  0.707388
 X  0.000000
 Y  0.000000
 Z  0.706825

to Rotation matrix
    0.000796  -1.000000   0.000000
    1.000000   0.000796   0.000000
    0.000000   0.000000   1.000000
```

Combuster54, 2024-09-12:
**Creating objects**

```cc
#include <stdio.h>
#include <gz/math/Quaternion.hh>
#include <iostream>
int main(int argc, char **argv)
{
    double w = 0.5;
    double x = 0.5;
    double y = 0.5;
    double z = 0.5;

    double v_roll = atan2(2.0 * (w * x + y * z), 1.0 - 2.0 * (x * x + y * y));

    double pitch = asin(2.0 * (w * y - z * x));

    double yaw = atan2(2.0 * (w * z + x * y), 1.0 - 2.0 * (y * y + z * z));

  gz::math::Quaterniond q(v_roll,pitch, yaw);
  gz::math::Matrix3d m(q);

  gz::math::Vector3d euler(q.Euler());
  printf(" W % .6f\n X % .6f\n Y % .6f\n Z % .6f\n", q.W(), q.X(), q.Y(), q.Z());
  std::cout << "\nto Rotation matrix\n";
  printf("   % .6f  % .6f  % .6f\n"
         "   % .6f  % .6f  % .6f\n"
         "   % .6f  % .6f  % .6f\n",
          m(0, 0), m(0, 1), m(0, 2),
          m(1, 0), m(1, 1), m(1, 2),
          m(2, 0), m(2, 1), m(2, 2));
}
```

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples>.\build\Release\object.exe
 W  0.500000
 X  0.500000
 Y  0.500000
 Z  0.500000

to Rotation matrix
    0.000000   0.000000   1.000000
    1.000000   0.000000   0.000000
    0.000000   1.000000   0.000000

(gz-ws) C:\Users\User\gazebo-ws\test\gz-math\examples
```

Works as expected

closed 2024-09-13 by azeey