gazebosim/gazebo_test_cases #1390: gz-msgs: Installation

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/msgs/11/install.html)
* [Source](https://github.com/gazebosim/gz-msgs/tree/gz-msgs11/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

sreekanthr, 2024-09-13:
I would like to pick this for testing in 
```
mac OS Monterey (12.7.6) [amd64]
Xcode 14.2 | Build version 14C18
```

sreekanthr, 2024-09-13:
I get the below error when installing version 1 as suggested in the tutorial

$ **brew install gz-msgs1**
```
Warning: You are using macOS 12.
We (and Apple) do not provide support for this old version.
It is expected behaviour that some formulae will fail to build in this old version.
It is expected behaviour that Homebrew will be buggy and slow.
Do not create any issues about this on Homebrew's GitHub repositories.
Do not create any issues even if you think this message is unrelated.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.

Warning: osrf/simulation/ignition-msgs1 has been deprecated because it is past end-of-life date! It was disabled on 2024-08-31.
==> Fetching osrf/simulation/ignition-msgs1
==> Downloading https://github.com/gazebosim/gz-msgs/commit/88386e4e7a38d0ccb0a96f9774ba5339bc7e5440.patch?full_index=1
Already downloaded: /Users/aromal/Library/Caches/Homebrew/downloads/297d7ca53ebb2cfff10164155e50f81e015af447f5f47e848fe993d297f771e1--88386e4e7a38d0ccb0a96f9774ba5339bc7e5440.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-msgs/releases/ignition-msgs-1.0.0.tar.bz2
Already downloaded: /Users/aromal/Library/Caches/Homebrew/downloads/dbe722ab40484e9aa3824ff84d5425f18f60d39e2072586f54505d0d01a528c2--ignition-msgs-1.0.0.tar.bz2
==> Installing ignition-msgs1 from osrf/simulation
==> Patching
==> Applying 88386e4e7a38d0ccb0a96f9774ba5339bc7e5440.patch
patching file include/ignition/msgs/Generator.hh
patching file src/Generator.cc
==> cmake .. -DBUILD_TESTING=Off -DCMAKE_INSTALL_RPATH=@loader_path/../lib
==> make install
Last 15 lines from /Users/aromal/Library/Logs/Homebrew/ignition-msgs1/02.make:
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [src/CMakeFiles/ign_msgs_gen.dir/generator_main.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
20 errors generated.
make[2]: *** [src/CMakeFiles/ign_msgs_gen.dir/Generator.cc.o] Error 1
make[1]: *** [src/CMakeFiles/ign_msgs_gen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Linking CXX static library libgtest.a
cd /tmp/ignition-msgs1-20240913-22304-6k5tkp/ignition-msgs-1.0.0/build/test && /usr/local/Cellar/cmake/3.30.3/bin/cmake -P CMakeFiles/gtest.dir/cmake_clean_target.cmake
cd /tmp/ignition-msgs1-20240913-22304-6k5tkp/ignition-msgs-1.0.0/build/test && /usr/local/Cellar/cmake/3.30.3/bin/cmake -E cmake_link_script CMakeFiles/gtest.dir/link.txt --verbose=1
/usr/bin/ar qc libgtest.a "CMakeFiles/gtest.dir/gtest/src/gtest-all.cc.o"
/usr/bin/ranlib libgtest.a
[  2%] Built target gtest
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/homebrew-core!

Error: You are using macOS 12.
We (and Apple) do not provide support for this old version.
It is expected behaviour that some formulae will fail to build in this old version.
It is expected behaviour that Homebrew will be buggy and slow.
Do not create any issues about this on Homebrew's GitHub repositories.
Do not create any issues even if you think this message is unrelated.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.

Do not report this issue: you are running in an unsupported configuration.
```

$ **brew install gz-msgs2**
```
Warning: No available formula with the name "gz-msgs2". Did you mean gz-msgs9, gz-msgs11 or gz-msgs10?
==> Searching for similarly named formulae and casks...
==> Formulae
osrf/simulation/gz-msgs9                                                        osrf/simulation/gz-msgs11 ✔                                                     osrf/simulation/gz-msgs10

To install osrf/simulation/gz-msgs9, run:
```

sreekanthr, 2024-09-13:
I already have the version 11 of gz-msg installed but I am unable to run it.
Getting the below error

$  **gz msg --versions**
``` 11.0.0~pre1```

$  **gz msg -l**
```
/usr/local/Cellar/gz-msgs11/11.0.0-pre1_2/lib/ruby/gz/cmdmsgs11.rb:33:in `execute': uninitialized constant Cmd::LIBRARY_NAME (NameError)
	from /usr/local/bin/gz:309:in `<main>'
```

 $  **gz msg --list**
```
/usr/local/Cellar/gz-msgs11/11.0.0-pre1_2/lib/ruby/gz/cmdmsgs11.rb:33:in `execute': uninitialized constant Cmd::LIBRARY_NAME (NameError)
	from /usr/local/bin/gz:309:in `<main>'
```

sreekanthr, 2024-09-13:
Created gz-msgs issue [#462](https://github.com/gazebosim/gz-msgs/issues/462)

closed 2024-09-13 by azeey