gazebosim/gazebo_test_cases #1392: gz-msgs: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/msgs/11/install.html)
* [Source](https://github.com/gazebosim/gz-msgs/tree/gz-msgs11/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-13:
**Binary Installation**

There's no msgs11 

```sh
(gz-ws) C:\Users\User>conda install libgz-msgs11 --channel conda-forge
Channels:
 - conda-forge
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: failed
PackagesNotFoundError: The following packages are not available from current channels:
  - libgz-msgs11
Current channels:
  - https://conda.anaconda.org/conda-forge
  - defaults
To search for alternate channels that may provide the conda package you're
looking for, navigate to
    https://anaconda.org
and use the search bar at the top of the page.
(gz-ws) C:\Users\User>.
```

**Source Instalation**

I wasn't able to install  `libgz-cmake4` `libgz-math8` `libgz-tools3` via `conda install`. So I have to build from source. Despite this, I found and error 

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-msgs\build>cmake .. -DBUILD_TESTING=OFF
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.41.34120.0
-- The CXX compiler identification is MSVC 19.41.34120.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.41.34120/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.41.34120/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-msgs11 version 11.0.0~pre1
-- Operating system is Windows
--

-- ====== Finding Dependencies ======
-- Found ZLIB: C:/Users/User/anaconda3/envs/gz-ws/Library/lib/z.lib (found version "1.3.1")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- Looking for Protobuf - found

-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [cli]
CMake Warning at C:/Users/User/anaconda3/envs/gz-ws/Library/share/cmake-3.30/Modules/CMakeFindDependencyMacro.cmake:76 (find_package):
  By not providing "Findgz-utils3-cli.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz-utils3-cli", but CMake did not find one.

  Could not find a package configuration file provided by "gz-utils3-cli"
  (requested version 3.0.0) with any of the following names:

    gz-utils3-cliConfig.cmake
    gz-utils3-cli-config.cmake

  Add the installation prefix of "gz-utils3-cli" to CMAKE_PREFIX_PATH or set
  "gz-utils3-cli_DIR" to a directory containing one of the above files.  If
  "gz-utils3-cli" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  C:/Program Files (x86)/gz-utils3/lib/cmake/gz-utils3/gz-utils3-config.cmake:191 (find_dependency)
  C:/Program Files (x86)/gz-cmake4/share/cmake/gz-cmake4/cmake4/GzFindPackage.cmake:201 (find_package)
  CMakeLists.txt:95 (gz_find_package)

CMake Warning at C:/Program Files (x86)/gz-cmake4/share/cmake/gz-cmake4/cmake4/GzFindPackage.cmake:201 (find_package):
  Found package configuration file:

    C:/Program Files (x86)/gz-utils3/lib/cmake/gz-utils3/gz-utils3-config.cmake

  but it set gz-utils3_FOUND to FALSE so package "gz-utils3" is considered to
  be NOT FOUND.  Reason given by package:

  gz-utils3 could not be found because dependency gz-utils3-cli could not be
  found.

Call Stack (most recent call first):
  CMakeLists.txt:95 (gz_find_package)

-- Looking for gz-utils3 - not found

-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Looking for gz-math8 - found

-- Found Python3: C:/Users/User/anaconda3/envs/gz-ws/python.exe (found version "3.11.10") found components: Interpreter
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 10.0.0
-- Looking for tinyxml2 - found

CMake Error at C:/Program Files (x86)/gz-cmake4/share/cmake/gz-cmake4/cmake4/GzConfigureBuild.cmake:63 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:142 (gz_configure_build)

CMake Error at C:/Program Files (x86)/gz-cmake4/share/cmake/gz-cmake4/cmake4/GzConfigureBuild.cmake:65 (message):
  --    Missing dependency [gz-utils3] (Components: cli)
Call Stack (most recent call first):
  CMakeLists.txt:142 (gz_configure_build)

CMake Error at C:/Program Files (x86)/gz-cmake4/share/cmake/gz-cmake4/cmake4/GzConfigureBuild.cmake:67 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:142 (gz_configure_build)

CMake Error at C:/Program Files (x86)/gz-cmake4/share/cmake/gz-cmake4/cmake4/GzConfigureBuild.cmake:72 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:142 (gz_configure_build)
-- Configuring incomplete, errors occurred!

(gz-ws) C:\Users\User\gazebo-ws\test\gz-msgs\build>
```

Combuster54, 2024-09-13:
## Update

**Solution**

Add the path of gz-utils3 

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-msgs\build>cmake .. -DCMAKE_PREFIX_PATH="C:/Program Files (x86)/gz-utils3"
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- gz-msgs11 version 11.0.0~pre1
-- Operating system is Windows
--

-- ====== Finding Dependencies ======
-- Looking for Protobuf - found

-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [cli]
-- Looking for gz-utils3-cli -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-cli
-- Looking for gz-utils3 - found

-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Looking for gz-math8 - found

-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 10.0.0
-- Looking for tinyxml2 - found

-- Configuring library: gz-msgs11
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- The program [cppcheck] was not found! Skipping codecheck setup
-- Build configuration successful
-- Build type: RelWithDebInfo
-- Install prefix: C:/Program Files (x86)/gz-msgs11
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Configuring done (2.7s)
-- Generating done (0.4s)
-- Build files have been written to: C:/Users/User/gazebo-ws/test/gz-msgs/build

(gz-ws) C:\Users\User\gazebo-ws\test\gz-msgs\build>cmake --build . --config Release
Versión de MSBuild 17.11.2+c078802d4 para .NET Framework

  1>Checking Build System
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-msgs/core/CMakeLists.txt
  Generator.cc
  generator_main.cc
  Generator.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command lin
  e to improve linker performance
  Generating code
  Finished generating code
  gz-msgs11_protoc_plugin.vcxproj -> C:\Users\User\gazebo-ws\test\gz-msgs\build\bin\Release\gz-msgs11_protoc_plugin.exe
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/actor.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/actuators.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/air_pressure_sensor.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/air_speed.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/air_speed_sensor.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/altimeter.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/altimeter_sensor.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/annotated_axis_aligned_2d_box.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/annotated_axis_aligned_2d_box_v.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/annotated_oriented_3d_box.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/annotated_oriented_3d_box_v.proto
  Running protoc on C:/Users/User/gazebo-ws/test/gz-msgs/proto/gz/msgs/any.proto
[... 448 lines cut ...]
```

Theres is a warning or error at the end

```sh
 register.cc
  Factory.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line
  to improve linker performance
     Creating library C:/Users/User/gazebo-ws/test/gz-msgs/build/lib/Release/gz-msgs11.lib and object C:/Users/User/gazebo-ws/test
  /gz-msgs/build/lib/Release/gz-msgs11.exp
  Generating code
C:\Users\User\anaconda3\envs\gz-ws\Library\include\google\protobuf\generated_message_tctable_decl.h(242): warning C4715: 'google::
protobuf::internal::MakeMapTypeCard': not all control paths return a value [C:\Users\User\gazebo-ws\test\gz-msgs\build\core\gz-msg
s11.vcxproj]
  Finished generating code
  gz-msgs11.vcxproj -> C:\Users\User\gazebo-ws\test\gz-msgs\build\bin\Release\gz-msgs11.dll
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-msgs/core/cmd/CMakeLists.txt
  msgs_main.cc
  msgs_main.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command lin
  e to improve linker performance
  Generating code
  Finished generating code
  gz-msgs.vcxproj -> C:\Users\User\gazebo-ws\test\gz-msgs\build\bin\Release\gz-msgs.exe
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-msgs/CMakeLists.txt
```

azeey, 2024-09-13:
I think that warning is coming from protobuf so not much we can do here.

closed 2024-09-13 by azeey