gazebosim/gazebo_test_cases #1403: gz-physics: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/8/installation.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/02_installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

shameekganguly, 2024-08-29:
I'd like to test this one!

Combuster54, 2024-09-09:
## Ubuntu 24.04 Noble

**Binari Install**

No errors or warnings everything ok.

**Source Install**

Warnings

```bash
combuster@combuster54:~/test$ cd gz-physics
mkdir build
cd build
cmake ..
make
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-physics8 version 8.0.0~pre2
-- Operating system is Linux
-- Found CPack generators: DEB
-- 

-- ====== Finding Dependencies ======
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [log]
-- Looking for gz-utils3-log -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-log
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'uuid'
--   Found uuid, version 2.39.3
-- Found UUID: TRUE  
-- Searching for <gz-common6> component [geospatial]
-- Looking for gz-common6-geospatial -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-geospatial
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-common6> component [graphics]
-- Looking for gz-common6-graphics -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-graphics
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Found GzAssimp: TRUE (found version "5.3.0") 
-- Searching for <gz-common6> component [graphics]
-- Searching for <gz-common6> component [profiler]
-- Looking for gz-common6-profiler -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-profiler
-- Looking for gz-common6 - found

-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
[... 102 lines cut ...]
-- Configuring library: gz-physics8-heightmap
-- Configuring library: gz-physics8-mesh
-- Skipping the component [dartsim] because the following packages are missing: DART
-- Configuring library: gz-physics8-tpelib
-- Adding 9 UNIT_tpelib tests
-- Configuring library: gz-physics8-tpe
-- Configuring library: gz-physics8-tpe-plugin
-- Adding 4 UNIT_tpe tests
-- Configuring library: gz-physics8-bullet
-- Configuring library: gz-physics8-bullet-plugin
-- Configuring library: gz-physics8-bullet-featherstone
-- Configuring library: gz-physics8-bullet-featherstone-plugin
-- The program [cppcheck] was not found! Skipping codecheck setup
-- Build configuration successful
-- Build type: RelWithDebInfo
-- Install prefix: /usr/local
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done (2.6s)
-- Generating done (0.1s)
-- Build files have been written to: /home/combuster/test/gz-physics/build
```

combuster@combuster54:~/test/gz-physics/build$ ls lib
lib/                                         libgz-physics-bullet-plugin.so
libgz-physics-bullet-featherstone-plugin.so  libgz-physics-tpe-plugin.so

azeey, 2024-09-09:
The DARTsim warning is relevant. That's the default physics engine so, this is a bug in either the install instructions or our package. Running `sudo apt-get build-dep -y libgz-physics8-dev` did not work for me. Can you confirm if it worked for you?

Combuster54, 2024-09-09:
Oh, is this okay?

```bash
combuster@combuster54:~$ sudo apt-get install libgz-physics8-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgz-physics8-dev is already the newest version (8.0.0~pre2-1~noble).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
combuster@combuster54:~$ sudo apt-get build-dep -y libgz-physics8-dev
Reading package lists... Done
E: You must put some 'deb-src' URIs in your sources.list
```

iche033, 2024-09-10:
It also didn't work for me and I get the same error msg. One option is to update the instructions to explicitly apt install the dependencies listed in [packages.apt](https://github.com/gazebosim/gz-physics/blob/gz-physics8/.github/ci/packages.apt)

Combuster54, 2024-09-10:
It seems llike I need some permissions

```bash
combuster@combuster54:~$ sudo apt update && xargs sudo apt install -y < /home/combuster/test/gz-physics/.github/ci/packages.apt
[sudo] password for combuster: 
Hit:1 https://download.docker.com/linux/ubuntu noble InRelease
Get:2 http://security.ubuntu.com/ubuntu noble-security InRelease [126 kB]      
Hit:3 http://pe.archive.ubuntu.com/ubuntu noble InRelease                      
Get:4 http://pe.archive.ubuntu.com/ubuntu noble-updates InRelease [126 kB]     
Get:5 http://packages.ros.org/ros2/ubuntu noble InRelease [4,670 B]            
Err:5 http://packages.ros.org/ros2/ubuntu noble InRelease                      
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Get:6 http://packages.osrfoundation.org/gazebo/ubuntu-prerelease noble InRelease [4,252 B]
Get:7 http://security.ubuntu.com/ubuntu noble-security/universe amd64 c-n-f Metadata [10.1 kB]
Hit:8 http://pe.archive.ubuntu.com/ubuntu noble-backports InRelease            
Get:9 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease [4,261 B]
Err:9 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease   
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 67170598AF249743
Get:10 http://pe.archive.ubuntu.com/ubuntu noble-updates/main amd64 Packages [475 kB]
Hit:11 https://cyberbotics.com/debian binary-amd64/ InRelease
Get:12 http://pe.archive.ubuntu.com/ubuntu noble-updates/universe amd64 Packages [351 kB]
Get:13 http://pe.archive.ubuntu.com/ubuntu noble-updates/universe amd64 c-n-f Metadata [13.9 kB]
Reading package lists... Done              
W: GPG error: http://packages.ros.org/ros2/ubuntu noble InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
E: The repository 'http://packages.ros.org/ros2/ubuntu noble InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 67170598AF249743
```

iche033, 2024-09-11:
hmm ROS 2 GPG error, you may have to run the steps to add the ROS 2 GPG key and the repository to your sources list again: https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debs.html

Combuster54, 2024-09-11:
Uhmmm I'm still looking an GPG error
```sh
combuster@combuster54:~$ sudo apt install software-properties-common
sudo add-apt-repository universe
[sudo] password for combuster: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
software-properties-common is already the newest version (0.99.48).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Adding component(s) 'universe' to all repositories.
Press [ENTER] to continue or Ctrl-c to cancel.
Get:1 https://download.docker.com/linux/ubuntu noble InRelease [48.8 kB]
Get:2 https://download.docker.com/linux/ubuntu noble/stable amd64 Packages [13.8 kB]
Get:3 http://security.ubuntu.com/ubuntu noble-security InRelease [126 kB]      
Hit:4 http://pe.archive.ubuntu.com/ubuntu noble InRelease                      
Get:5 http://pe.archive.ubuntu.com/ubuntu noble-updates InRelease [126 kB]     
Get:6 http://packages.osrfoundation.org/gazebo/ubuntu-prerelease noble InRelease [4,252 B]
Get:7 http://security.ubuntu.com/ubuntu noble-security/main amd64 Packages [350 kB]
Get:8 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease [4,261 B]
Get:9 http://packages.ros.org/ros2/ubuntu noble InRelease [4,670 B]            
Err:8 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease   
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 67170598AF249743
Err:9 http://packages.ros.org/ros2/ubuntu noble InRelease                     
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Get:10 http://security.ubuntu.com/ubuntu noble-security/main Translation-en [77.1 kB]
Get:11 http://security.ubuntu.com/ubuntu noble-security/main amd64 c-n-f Metadata [4,416 B]
Get:12 http://security.ubuntu.com/ubuntu noble-security/restricted amd64 Packages [317 kB]
Get:13 http://security.ubuntu.com/ubuntu noble-security/restricted Translation-en [61.5 kB]
Get:14 http://security.ubuntu.com/ubuntu noble-security/universe amd64 Packages [267 kB]
Get:15 http://security.ubuntu.com/ubuntu noble-security/universe Translation-en [111 kB]
Get:16 http://security.ubuntu.com/ubuntu noble-security/universe amd64 c-n-f Metadata [10.1 kB]
Hit:17 http://pe.archive.ubuntu.com/ubuntu noble-backports InRelease           
Hit:18 https://cyberbotics.com/debian binary-amd64/ InRelease
Get:19 http://pe.archive.ubuntu.com/ubuntu noble-updates/main amd64 Packages [501 kB]
Get:20 http://pe.archive.ubuntu.com/ubuntu noble-updates/main Translation-en [123 kB]
Get:21 http://pe.archive.ubuntu.com/ubuntu noble-updates/main amd64 c-n-f Metadata [8,232 B]
Get:22 http://pe.archive.ubuntu.com/ubuntu noble-updates/restricted amd64 Packages [317 kB]
Get:23 http://pe.archive.ubuntu.com/ubuntu noble-updates/restricted Translation-en [61.5 kB]
Get:24 http://pe.archive.ubuntu.com/ubuntu noble-updates/universe amd64 Packages [353 kB]
Get:25 http://pe.archive.ubuntu.com/ubuntu noble-updates/universe Translation-en [145 kB]
Get:26 http://pe.archive.ubuntu.com/ubuntu noble-updates/universe amd64 c-n-f Metadata [13.9 kB]
Reading package lists... Done                                     
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 67170598AF249743
W: GPG error: http://packages.ros.org/ros2/ubuntu noble InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
E: The repository 'http://packages.ros.org/ros2/ubuntu noble InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
combuster@combuster54:~$ sudo apt update && sudo apt install curl -y
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
Hit:1 https://download.docker.com/linux/ubuntu noble InRelease
Hit:2 http://security.ubuntu.com/ubuntu noble-security InRelease               
Hit:3 http://pe.archive.ubuntu.com/ubuntu noble InRelease                      
Hit:4 http://pe.archive.ubuntu.com/ubuntu noble-updates InRelease              
Get:5 http://packages.ros.org/ros2/ubuntu noble InRelease [4,670 B]            
Err:5 http://packages.ros.org/ros2/ubuntu noble InRelease                      
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Hit:6 http://pe.archive.ubuntu.com/ubuntu noble-backports InRelease            
Get:7 http://packages.osrfoundation.org/gazebo/ubuntu-prerelease noble InRelease [4,252 B]
Get:8 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease [4,261 B]
Err:8 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 67170598AF249743
Hit:9 https://cyberbotics.com/debian binary-amd64/ InRelease
Reading package lists... Done
W: GPG error: http://packages.ros.org/ros2/ubuntu noble InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
[... cut ...]
W: Failed to fetch http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/noble/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 67170598AF249743
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (8.5.0-2ubuntu10.3).
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.
combuster@combuster54:~$ sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
combuster@combuster54:~$ 
```

azeey, 2024-09-11:
packages.osrfoundation.org needs a different key. Can you try the following?

```
sudo curl https://packages.osrfoundation.org/gazebo.gpg --output /usr/share/keyrings/pkgs-osrf-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-stable.list > /dev/null
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-prerelease $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-prerelease.list > /dev/null
sudo apt-get update
```

Combuster54, 2024-09-11:
Sure.

## Results 
```sh
combuster@combuster54:~$ sudo curl https://packages.osrfoundation.org/gazebo.gpg --output /usr/share/keyrings/pkgs-osrf-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-stable.list > /dev/null
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-prerelease $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-prerelease.list > /dev/null
sudo apt-get update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1222  100  1222    0     0   1794      0 --:--:-- --:--:-- --:--:--  1797
Hit:1 https://download.docker.com/linux/ubuntu noble InRelease
Hit:2 http://packages.ros.org/ros2/ubuntu noble InRelease                      
Hit:3 http://pe.archive.ubuntu.com/ubuntu noble InRelease                      
Get:4 http://packages.osrfoundation.org/gazebo/ubuntu-prerelease noble InRelease [4,252 B]
Hit:5 http://security.ubuntu.com/ubuntu noble-security InRelease               
Get:6 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease [4,261 B]
Hit:7 http://pe.archive.ubuntu.com/ubuntu noble-updates InRelease              
Hit:8 http://pe.archive.ubuntu.com/ubuntu noble-backports InRelease
Hit:9 https://cyberbotics.com/debian binary-amd64/ InRelease
Fetched 8,513 B in 1s (8,867 B/s)
Reading package lists... Done
combuster@combuster54:~$  sudo apt update && xargs sudo apt install -y < /home/combuster/test/gz-physics/.github/ci/packages.apt
Hit:1 https://download.docker.com/linux/ubuntu noble InRelease
Hit:2 http://security.ubuntu.com/ubuntu noble-security InRelease               
Hit:3 http://pe.archive.ubuntu.com/ubuntu noble InRelease                      
Hit:4 http://pe.archive.ubuntu.com/ubuntu noble-updates InRelease              
Get:5 http://packages.osrfoundation.org/gazebo/ubuntu-prerelease noble InRelease [4,252 B]
Hit:6 http://packages.ros.org/ros2/ubuntu noble InRelease                      
Hit:7 http://pe.archive.ubuntu.com/ubuntu noble-backports InRelease 
Get:8 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease [4,261 B]
Hit:9 https://cyberbotics.com/debian binary-amd64/ InRelease
Fetched 8,513 B in 1s (8,927 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
15 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libeigen3-dev is already the newest version (3.4.0-4).
libeigen3-dev set to manually installed.
libgz-cmake4-dev is already the newest version (4.0.0~pre1-1~noble).
libgz-cmake4-dev set to manually installed.
libgz-common6-dev is already the newest version (6.0.0~pre2-1~noble).
libgz-common6-dev set to manually installed.
libgz-math8-dev is already the newest version (8.0.0~pre1-1~noble).
libgz-math8-dev set to manually installed.
libgz-math8-eigen3-dev is already the newest version (8.0.0~pre1-1~noble).
libgz-math8-eigen3-dev set to manually installed.
libgz-plugin3-dev is already the newest version (3.0.0~pre1-1~noble).
libgz-plugin3-dev set to manually installed.
libgz-utils3-cli-dev is already the newest version (3.0.0~pre3-1~noble).
libgz-utils3-cli-dev set to manually installed.
libgz-utils3-dev is already the newest version (3.0.0~pre3-1~noble).
libgz-utils3-dev set to manually installed.
libsdformat15-dev is already the newest version (15.0.0~pre2-1~noble).
libsdformat15-dev set to manually installed.
The following additional packages will be installed:
  libbenchmark1.8.3 libccd-dev libconsole-bridge-dev libconsole-bridge1.0
  libdart-utils-urdf6.13 libdart-utils6.13
  libdart6.13-external-convhull-3d-dev libdart6.13-utils-dev libfcl-dev
  liboctomap-dev libode-dev libtinyxml-dev libtinyxml2.6.2v5 liburdfdom-dev
  liburdfdom-headers-dev liburdfdom-model-state4.0 liburdfdom-model4.0
  liburdfdom-sensor4.0 liburdfdom-world4.0
Suggested packages:
  libbenchmark-tools libtinyxml-doc
The following NEW packages will be installed:
  libbenchmark-dev libbenchmark1.8.3 libccd-dev libconsole-bridge-dev
  libconsole-bridge1.0 libdart-utils-urdf6.13 libdart-utils6.13
[... 176 lines cut ...]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
15 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libbenchmark-dev is already the newest version (1.8.3-3).
libdart6.13-collision-bullet-dev is already the newest version (6.13.2+ds1-1~osrf2~noble).
libdart6.13-collision-ode-dev is already the newest version (6.13.2+ds1-1~osrf2~noble).
libdart6.13-dev is already the newest version (6.13.2+ds1-1~osrf2~noble).
libdart6.13-external-ikfast-dev is already the newest version (6.13.2+ds1-1~osrf2~noble).
libdart6.13-external-odelcpsolver-dev is already the newest version (6.13.2+ds1-1~osrf2~noble).
libdart6.13-utils-urdf-dev is already the newest version (6.13.2+ds1-1~osrf2~noble).
libeigen3-dev is already the newest version (3.4.0-4).
libgz-cmake4-dev is already the newest version (4.0.0~pre1-1~noble).
[... cut ...] oble).
libgz-math8-dev is already the newest version (8.0.0~pre1-1~noble).
libgz-math8-eigen3-dev is already the newest version (8.0.0~pre1-1~noble).
libgz-plugin3-dev is already the newest version (3.0.0~pre1-1~noble).
libgz-utils3-cli-dev is already the newest version (3.0.0~pre3-1~noble).
libgz-utils3-dev is already the newest version (3.0.0~pre3-1~noble).
libsdformat15-dev is already the newest version (15.0.0~pre2-1~noble).
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.
```

Combuster54, 2024-09-12:
**Final Result**

```sh
combuster@combuster54:~/test_physics/gz-physics/build$ cmake ..
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-physics8 version 8.0.0~pre2
-- Operating system is Linux
-- Found CPack generators: DEB
-- 

-- ====== Finding Dependencies ======
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [log]
-- Looking for gz-utils3-log -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-log
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'uuid'
--   Found uuid, version 2.39.3
-- Found UUID: TRUE  
-- Searching for <gz-common6> component [geospatial]
-- Looking for gz-common6-geospatial -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-geospatial
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-common6> component [graphics]
-- Looking for gz-common6-graphics -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-graphics
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Found GzAssimp: TRUE (found version "5.3.0") 
-- Searching for <gz-common6> component [graphics]
-- Searching for <gz-common6> component [profiler]
-- Looking for gz-common6-profiler -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-profiler
-- Looking for gz-common6 - found

-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-common6> component [testing]
-- Looking for gz-common6-testing -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-testing
-- Looking for gz-common6 - found

-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-math8> component [eigen3]
-- Looking for gz-math8-eigen3 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8-eigen3
-- Looking for gz-math8 - found
[... 334 lines cut ...]
[ 89%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/EntityManagementFeatures.cc.o
[ 90%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/FreeGroupFeatures.cc.o
[ 90%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/JointFeatures.cc.o
[ 91%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/KinematicsFeatures.cc.o
[ 91%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/SDFFeatures.cc.o
[ 92%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/ShapeFeatures.cc.o
[ 92%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/SimulationFeatures.cc.o
[ 93%] Building CXX object bullet/CMakeFiles/gz-physics8-bullet-plugin.dir/src/plugin.cc.o
[ 93%] Linking CXX shared library ../lib/libgz-physics8-bullet-plugin.so
[ 93%] Built target gz-physics8-bullet-plugin
[ 93%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/Base.cc.o
[ 94%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/EntityManagementFeatures.cc.o
[ 94%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/FreeGroupFeatures.cc.o
[ 95%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/JointFeatures.cc.o
[ 95%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/KinematicsFeatures.cc.o
[ 96%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/LinkFeatures.cc.o
[ 96%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/SDFFeatures.cc.o
[ 97%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/ShapeFeatures.cc.o
[ 97%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/SimulationFeatures.cc.o
[ 98%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/WorldFeatures.cc.o
[ 98%] Building CXX object bullet-featherstone/CMakeFiles/gz-physics8-bullet-featherstone-plugin.dir/src/plugin.cc.o
[100%] Linking CXX shared library ../lib/libgz-physics8-bullet-featherstone-plugin.so
[100%] Built target gz-physics8-bullet-featherstone-plugin
combuster@combuster54:~/test_physics/gz-physics/build$ 
```

closed 2024-09-13 by azeey