gazebosim/gazebo_test_cases #1409: gz-physics: Switching physics engines

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/8/switchphysicsengines.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/04-switching-physics-engines.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

shameekganguly, 2024-08-29:
I'd like to test this one!

Combuster54, 2024-09-09:
## Ubuntu 24.04 Noble

**From SDF**

```
<?xml version="1.0" ?>
<sdf version="1.7">
  <world name="custom_world">

    <plugin
      filename="gz-sim-physics-system"
      name="gz::sim::systems::Physics">
      <engine>
        <filename>HelloWorldPlugin</filename>
      </engine>
    </plugin>

    <light name="sun" type="directional">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 -0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.5 -1</direction>
    </light>

  </world>
</sdf>
```
bash
```bash
combuster@combuster54:~$ gz sim /home/combuster/test/gazebo-ionic-physics/world/physic_plugin_world.sdf
(2024-09-09 12:57:25.792) [error] No physics plugins found in library [/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libHelloWorldPlugin.so].
```
[image]

error related in: https://github.com/gazebosim/gz-sim/pull/2604

**From the command line**

[image]

**From C++ API** 

Link is not working, I can't test. Any idea?

[image]

ahcorde, 2024-09-10:
PR https://github.com/gazebosim/gz-physics/pull/682

shameekganguly, 2024-09-13:
I've removed this tutorial in https://github.com/gazebosim/gz-physics/pull/684 since it is not linked directly on https://gazebosim.org/api/physics/8/tutorials.html.

shameekganguly, 2024-09-13:
Thanks anyway @Combuster54 for testing the integration with gz-sim 🙇 (I noticed that you also tested the related gz-sim tutorial in https://github.com/gazebosim/gazebo_test_cases/issues/1260, thanks!)

closed 2024-09-13 by shameekganguly