gazebosim/gazebo_test_cases #1412: gz-physics: Use different physics engines

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/8/physicsengine.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/04_physics_engines.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

shameekganguly, 2024-08-29:
I'd like to test this one!

Combuster54, 2024-09-10:
## Ubuntu 24.04 Noble

Links are not working

Combuster54, 2024-09-11:
@j-rivero So it's the same as https://github.com/gazebosim/gazebo_test_cases/issues/1409#issuecomment-2339962435

then what's the difference here? those issues are the same then

shameekganguly, 2024-09-13:
I'm updating the link in https://github.com/gazebosim/gz-physics/pull/684 and removing the other tutorial which is not explicitly referenced from the [gz-physics tutorials page](https://gazebosim.org/api/physics/8/tutorials.html).

closed 2024-09-13 by shameekganguly