gazebosim/gazebo_test_cases #1415: gz-physics: Plugin Loading

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/8/pluginloading.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/05_plugin_loading.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

shameekganguly, 2024-08-29:
I'd like to test this one!

Combuster54, 2024-09-10:
## Ubuntu 24.04  Noble

**Compile the loader**

```sh
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ cmake ..
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for gz-plugin3 -- found version 3.0.0~pre1
-- Searching for dependencies of gz-plugin3
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [cli]
-- Looking for gz-utils3-cli -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-cli
-- Searching for <gz-plugin3> component [all]
-- Looking for all libraries of gz-plugin3 -- found version 3.0.0~pre1
-- Looking for gz-plugin3 -- found version 3.0.0~pre1
-- Looking for gz-plugin3-loader -- found version 3.0.0~pre1
-- Searching for dependencies of gz-plugin3-loader
-- Looking for gz-plugin3-register -- found version 3.0.0~pre1
-- Searching for dependencies of gz-plugin3-register
-- Looking for gz-physics8 -- found version 8.0.0~pre2
-- Searching for dependencies of gz-physics8
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [log]
-- Looking for gz-utils3-log -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-log
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'uuid'
--   Found uuid, version 2.39.3
-- Found UUID: TRUE  
-- Searching for <gz-common6> component [testing]
-- Looking for gz-common6-testing -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-testing
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-math8> component [eigen3]
-- Looking for gz-math8-eigen3 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8-eigen3
-- Looking for gz-plugin3 -- found version 3.0.0~pre1
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Configuring done (0.4s)
-- Generating done (0.0s)
-- Build files have been written to: /home/combuster/test/gazebo-ionic-physics/hello_world_loader/build
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ make
[ 50%] Building CXX object CMakeFiles/hello_world_loader.dir/hello_world_loader.cc.o
[100%] Linking CXX executable hello_world_loader
[100%] Built target hello_world_loader
```

**Load existing plugins**

```sh
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ ./hello_world_loader /usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so
Testing plugin: gz::physics::dartsim::Plugin
  engine name: dartsim-6.13.2
```

Works as expected, but it's another version of dartsim. The tutorial shows dartsim-6.10.0

azeey, 2024-09-13:
Mind creating a PR for the dartsim version? I'll close this since a fairly minor issue.

closed 2024-09-13 by azeey

Combuster54, 2024-09-14:
Sure @azeey. I'm going to read the contribute guide

Combuster54, 2024-09-14:
PR https://github.com/gazebosim/gz-physics/pull/686