gazebosim/gazebo_test_cases #1418: gz-physics: Physics simulation concepts

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/8/physicsconcepts.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/06-physics-simulation-concepts.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

shameekganguly, 2024-08-29:
I'd like to test this one!

Combuster54, 2024-09-10:
## Ubuntu 24.04 Noble

**Prerequisite**

[Use different physics engines](https://gazebosim.org/api/physics/8/physicsengine.html) : The page is not found.

Combuster54, 2024-09-10:
**Physics concepts in Gazebo simulation**
 
**Differential Drive**
```sh
combuster@combuster54:~$ wget https://raw.githubusercontent.com/gazebosim/gz-sim/gz-physics8/examples/worlds/diff_drive.sdf -P ~
--2024-09-10 12:10:41--  https://raw.githubusercontent.com/gazebosim/gz-sim/gz-physics8/examples/worlds/diff_drive.sdf
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-09-10 12:10:41 ERROR 404: Not Found.

combuster@combuster54:~$ wget https://raw.githubusercontent.com/gazebosim/gz-sim/gz-physics8/examples/worlds/velocity_control.sdf -P ~
--2024-09-10 12:10:47--  https://raw.githubusercontent.com/gazebosim/gz-sim/gz-physics8/examples/worlds/velocity_control.sdf
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-09-10 12:10:47 ERROR 404: Not Found.
```

Combuster54, 2024-09-10:
## Using default dartsim

**Monitor the model and link pose**

[image]

**Collision effect**
[image]

## Using TPE

[image]

I think we might be use the same sdf

`velocity_control.sdf`
[image]
`diff_drive.sdf`
[image]

**Error** : `(2024-09-10 12:27:48.785) [error] Attempting to create a new joint [left_wheel_joint] but the chosen physics engine does not support the minimal joint features, so no joints will be created.`
```
combuster@combuster54:~$ gz sim diff_drive.sdf --physics-engine gz-physics-tpe-plugin # specify TPE as physics engine
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
(2024-09-10 12:27:40.039) [error] Attempting to create a new joint [left_wheel_joint] but the chosen physics engine does not support the minimal joint features, so no joints will be created.
combuster@combuster54:~$ gz sim velocity_control.sdf --physics-engine gz-physics-tpe-plugin # specify TPE as physics engine
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
(2024-09-10 12:27:48.785) [error] Attempting to create a new joint [left_wheel_joint] but the chosen physics engine does not support the minimal joint features, so no joints will be created.
```

Combuster54, 2024-09-10:
**Investigation**
The error is confuse. Looking the syntax inside `left_wheel_joint` . I can say that is the same as the other wheel_joints, even the wheel_links.

azeey, 2024-09-13:
TPE does not support joints, so that's expected

closed 2024-09-13 by azeey