gazebosim/gazebo_test_cases #1424: gz-physics: Implement a custom feature

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/8/createcustomfeature.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/08-implementing-a-custom-feature.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

shameekganguly, 2024-08-29:
I'd like to test this one!

Combuster54, 2024-09-10:
**Folder structure of the plugins**

[image]

dartsim folder doesn't have include folder, everything is inside src
[image]

**Plugin and feature requirements**

FindFreeGroupFeature doesn't have a link, is this correct?

[image]

**Implement the custom feature**

CustomFeatures.hh and cc has the same sintax as the tutorial. 

is it a way to test it?

Combuster54, 2024-09-10:
In order to follow the structure of the first photo. I change this line in CMakeLists.txt

```
target_include_directories(${features} SYSTEM INTERFACE ${DART_INCLUDE_DIRS})
```
to 
```
target_include_directories(${features} SYSTEM INTERFACE
  ${DART_INCLUDE_DIRS}
  ${CMAKE_CURRENT_SOURCE_DIR}/include/gz/physics/dartsim
)
```
then I move all the headers into the correct folder 
[image]

Finally, I need to change includes to follow the new path

example:

```c++
#include "CustomFeatures.hh"
```
to

```c++
#include "gz/physics/dartsim/CustomFeatures.hh"
```

results

```sh
combuster@combuster54:~/test/gz-physics/build$ cmake ..
-- gz-physics8 version 8.0.0~pre2
-- Operating system is Linux
-- Found CPack generators: DEB
-- 

-- ====== Finding Dependencies ======
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [log]
-- Looking for gz-utils3-log -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-log
-- Checking for module 'uuid'
--   Found uuid, version 2.39.3
-- Searching for <gz-common6> component [geospatial]
-- Looking for gz-common6-geospatial -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-geospatial
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-common6> component [graphics]
-- Looking for gz-common6-graphics -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-graphics
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for <gz-common6> component [graphics]
-- Searching for <gz-common6> component [profiler]
-- Looking for gz-common6-profiler -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6-profiler
-- Looking for gz-common6 - found

-- Looking for gz-common6 -- found version 6.0.0~pre2
-- Searching for dependencies of gz-common6
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for <gz-common6> component [testing]
[... 151 lines cut ...]
[ 51%] Built target INTEGRATION_JointTypes2f
[ 52%] Built target INTEGRATION_JointTypes3d
[ 54%] Built target INTEGRATION_JointTypes3f
[ 55%] Built target INTEGRATION_RequestFeatures
[ 56%] Built target PERFORMANCE_ExpectData
[ 58%] Built target PERFORMANCE_symbol_names
[ 65%] Built target gz-physics8-tpelib
[ 66%] Built target UNIT_tpelib_AABBTree_TEST
[ 68%] Built target UNIT_tpelib_CollisionDetector_TEST
[ 69%] Built target UNIT_tpelib_Collision_TEST
[ 70%] Built target UNIT_tpelib_Engine_TEST
[ 71%] Built target UNIT_tpelib_Link_TEST
[ 72%] Built target UNIT_tpelib_Model_TEST
[ 73%] Built target UNIT_tpelib_Shape_TEST
[ 76%] Built target UNIT_tpelib_Utils_TEST
[ 77%] Built target UNIT_tpelib_World_TEST
[ 83%] Built target gz-physics8-tpe-plugin
[ 84%] Built target UNIT_tpe_Base_TEST
[ 85%] Built target UNIT_tpe_EntityManagement_TEST
[ 86%] Built target UNIT_tpe_SDFFeatures_TEST
[ 87%] Built target UNIT_tpe_SimulationFeatures_TEST
[ 92%] Built target gz-physics8-bullet-plugin
[100%] Built target gz-physics8-bullet-featherstone-plugin
```
is it okay?

iche033, 2024-09-14:
I think this tutorial is quite outdated. Created https://github.com/gazebosim/gz-physics/issues/685

closed 2024-09-14 by iche033