gazebosim/gazebo_test_cases #1428: gz-physics: Use a custom engine with Gazebo Physics

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/physics/8/usecustomengine.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics8/tutorials/09_use_custom_engine.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Sanjay1911, 2024-08-30:
I would like to try this

Sanjay1911, 2024-09-01:
My windows installation didn't work as expected. Switching to Ubuntu Noble for testing

azeey, 2024-09-03:
@Sanjay1911 Did you face any problems that we can try to figure out when installing on windows?

Sanjay1911, 2024-09-04:
It was the same installation issue that you solved in #1701 
But I got my Ubuntu Noble setup working.

azeey, 2024-09-04:
Got it. If you do have time to test windows though, it would be greatly appreciated. In general, we don't have many folks testing on Windows.

peci1, 2024-09-16:
I've re-worked the build instructions in https://github.com/gazebosim/gz-physics/pull/689 .

With that change, this issue can be closed.

closed 2024-09-16 by azeey