gazebosim/gazebo_test_cases #1436: gz-plugin: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/plugin/3/installation.html)
* [Source](https://github.com/gazebosim/gz-plugin/tree/gz-plugin3/tutorials/02_installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-09-10:
## OS: Ubuntu 24.04 (Noble)
### Binary Install
Everything works fine

### Source Install
Packages libgz-tools3-dev and gz-tools3 are not available upstream, instead version 2 of those packages is available.

I tested gz-plugin build against libgz-tools2-dev and gz-tools2

### Documentation
Everything works fine

Note: There was **api.dox** has become obsolete warning but it didn't affect the documentation generated.

### Testing
I had following issue with making coverage:
[image]

iche033, 2024-09-11:
thanks created a PR to update the gz-tools version in https://github.com/gazebosim/gz-plugin/pull/153

closed 2024-09-13 by azeey