gazebosim/gazebo_test_cases #1437: gz-plugin: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/plugin/3/installation.html)
* [Source](https://github.com/gazebosim/gz-plugin/tree/gz-plugin3/tutorials/02_installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-13:
## Windows 11

**Binary Instalation**

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-common\build>conda install libgz-plugin3 --channel conda-forge
Channels:
 - conda-forge
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - libgz-plugin3

Current channels:

  - https://conda.anaconda.org/conda-forge
  - defaults

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

```

**Source Instalation**

I use `cmake .. -DBUILD_TESTIN=OFF -DCMAKE_PREFIX_PATH="C:/Program Files (x86)/gz-utils3"` to solve the problem with gz-utils3-cli

**Without DCMAKE_PREFIX_PATH**

```sh
(gz-ws) C:\Users\User\gazebo-ws\test\gz-plugin\build>cmake .. -DBUILD_TESTIN=OFF
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.41.34120.0
-- The CXX compiler identification is MSVC 19.41.34120.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.41.34120/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.41.34120/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-plugin3 version 3.0.0~pre1
-- Operating system is Windows
--

-- ====== Finding Dependencies ======
-- Found DL: TRUE
-- Looking for libdl - found

-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [cli]
CMake Warning at C:/Users/User/anaconda3/envs/gz-ws/Library/share/cmake-3.30/Modules/CMakeFindDependencyMacro.cmake:76 (find_package):
  By not providing "Findgz-utils3-cli.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz-utils3-cli", but CMake did not find one.

  Could not find a package configuration file provided by "gz-utils3-cli"
[... 303 lines cut ...]
  gz_TEST.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line to
  improve linker performance
  Generating code
  Finished generating code
  UNIT_gz_TEST.vcxproj -> C:\Users\User\gazebo-ws\test\gz-plugin\build\bin\Release\UNIT_gz_TEST.exe
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-plugin/core/CMakeLists.txt
  utility_TEST.cc
  utility_TEST.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command lin
  e to improve linker performance
  Generating code
  Finished generating code
  UNIT_utility_TEST.vcxproj -> C:\Users\User\gazebo-ws\test\gz-plugin\build\bin\Release\UNIT_utility_TEST.exe
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-plugin/loader/src/cmd/CMakeLists.txt
  gz.cc
  gz.vcxproj -> C:\Users\User\gazebo-ws\test\gz-plugin\build\lib\Release\gz.lib
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-plugin/loader/src/cmd/CMakeLists.txt
  plugin_main.cc
  plugin_main.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line
   to improve linker performance
  Generating code
  Finished generating code
  gz-plugin.vcxproj -> C:\Users\User\gazebo-ws\test\gz-plugin\build\bin\Release\gz-plugin.exe
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-plugin/CMakeLists.txt

```

azeey, 2024-09-13:
There are two other options:
* Use colcon (see https://github.com/gazebosim/sdformat/pull/1481)
* Install the gazebo packages to the same directory using `-DCMAKE_INSTALL_PREFIX`

I'll go ahead and close this issue since all our windows tutorials have this problem with prereleases.

closed 2024-09-13 by azeey