gazebosim/gazebo_test_cases #1439: gz-rendering: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/rendering)
* [Source](https://github.com/gazebosim/gz-rendering/blob/gz-rendering9/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-01:
Interested in !

avanmalleghem, 2024-09-01:
"Please see [CONTRIBUTING.md](https://gazebosim.org/docs/all/contributing)." -> correct link but incorrect filename (should be "contribution guide")

Will create a PR soon.

avanmalleghem, 2024-09-02:
PR opened : https://github.com/gazebosim/gz-rendering/pull/1051

ahcorde, 2024-09-03:
Thank you for testing!

closed 2024-09-03 by ahcorde