gazebosim/gazebo_test_cases #1444: gz-rendering: Installation

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/rendering/9/installation.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/02_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

utkuesen, 2024-09-14:
Hi Dev Team,
I tried both binary and source installations.
I couldn't install ROS on MacOS by trying both of the ways :(
**I got this error at binary install:**
-------------------------------- Start of the terminal ----------------------------
MacBook-Pro ~ % ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

curl: (56) The requested URL returned error: 404
-------------------------------- End of the terminal ----------------------------

**I got these errors at source install:**
-------------------------------- Start of the terminal ----------------------------
git clone https://github.com/gazebosim/gz-rendering -b gz-rendering9
Cloning into 'gz-rendering'...
remote: Enumerating objects: 31410, done.
remote: Counting objects: 100% (2406/2406), done.
remote: Compressing objects: 100% (1429/1429), done.
remote: Total 31410 (delta 1284), reused 1958 (delta 967), pack-reused 29004 (from 1)
Receiving objects: 100% (31410/31410), 128.82 MiB | 4.55 MiB/s, done.
Resolving deltas: 100% (22602/22602), done.

brew install --only-dependencies gz-rendering9
Warning: No available formula with the name "gz-rendering9".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for gz-rendering9.

mkdir build
cd build

 cmake ..          

-- The C compiler identification is AppleClang 15.0.0.15000309
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:12 (find_package):
  By not providing "Findgz-cmake4.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "gz-cmake4", but CMake did not find one.

  Could not find a package configuration file provided by "gz-cmake4" with
  any of the following names:

    gz-cmake4Config.cmake
    gz-cmake4-config.cmake

  Add the installation prefix of "gz-cmake4" to CMAKE_PREFIX_PATH or set
  "gz-cmake4_DIR" to a directory containing one of the above files.  If
  "gz-cmake4" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
-------------------------------- End of the terminal ----------------------------
If you can say me something different to try, I can try it.
Best regards.

iche033, 2024-09-16:
checking if you ran this step: `brew tap osrf/simulation`?