gazebosim/gazebo_test_cases #1445: gz-rendering: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/9/installation.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/02_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-29:
I am interested in testing it out.

avanmalleghem, 2024-09-09:
Can I help on this one @trushant05 ?

trushant05, 2024-09-09:
## OS: Ubuntu 24.04 (Noble)
### Binary Install
Everything works fine

### Source Install
Everything works fine

### Documentation
Everything works fine

Note: There was **api.dox** has become obsolete warning but it didn't affect the documentation generated.

### Testing
I did run all the tests and everything worked out but I did receive following two tests which failed. I believe it is because my current system does not have external GPU. 

[image]

> Can I help on this one @trushant05 ?
@avanmalleghem if you have a system with external GPU can you check if this is test failure or my system lacking external GPU?

avanmalleghem, 2024-09-10:
Don't have the same output...
[image]

avanmalleghem, 2024-09-10:
Don't have the same output
[image]

iche033, 2024-09-14:
I can reproduce the failure. Created https://github.com/gazebosim/gz-rendering/issues/1057

closed 2024-09-14 by iche033