gazebosim/gazebo_test_cases #1446: gz-rendering: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/rendering/9/installation.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/02_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-11:
Interested in !

avanmalleghem, 2024-09-11:
Blocked by same kind of issue than https://github.com/gazebosim/gazebo_test_cases/issues/1446

avanmalleghem, 2024-09-13:
- I have the following output during `cmake .. -DBUILD_TESTING=OFF` but the compilation works !
[image]
- I opened a PR : https://github.com/gazebosim/gz-rendering/pull/1055

closed 2024-09-13 by azeey