gazebosim/gazebo_test_cases #1451: gz-rendering: Creating and using light maps

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/9/lightmap.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/04_lightmap.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-09-05:
[image]

### Not able to run,  giving this error

iche033, 2024-09-05:
it's trying to download the [indoor lightmap](https://app.gazebosim.org/OpenRobotics/fuel/models/Indoor%20lightmap) from Fuel (~97MB). Once downloaded it should be stored in the cache: `~/.gz/fuel/fuel.gazebosim.org/openrobotics/models/indoor lightmap`. Maybe it's still downloading?

akky20, 2024-09-07:
## Working Properly
[image]

closed 2024-09-09 by iche033