gazebosim/gazebo_test_cases #1460: gz-rendering: Custom scene viewer

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/9/custom_scene_viewer.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/13_custom_scene_viewer.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

akky20, 2024-09-03:
## I have tested it and running fine 
[image]

azeey, 2024-09-03:
@Mechazo11 are you going to test this? If so you will get the credit since you were the one assigned.

@akky20 I'll assign it to you if @Mechazo11 does not want to test this.

Mechazo11, 2024-09-04:
@azeey please credit @akky20 for it.

iche033, 2024-09-04:
updated assignee

closed 2024-09-04 by iche033