gazebosim/gazebo_test_cases #1463: gz-rendering: Camera tracking

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/9/camera_tracking.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/14_camera_tracking_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

csimon-be, 2024-08-30:
Interested in this one

Combuster54, 2024-09-09:
## Ubuntu 24.04 Noble

**Compile and run the example**

```bash
combuster@combuster54:~/test/gz-rendering/examples/camera_tracking/build$ ./camera_tracking
(2024-09-09 11:04:03.804) [info] Loading plugin [gz-rendering-ogre]
===============================
  TAB - Switch render engines  
  ESC - Exit                   
                               
  W: Move box forward          
  S: Move box backward         
  A: Rotate box to the left    
  D: Rotate box to the right   
                               
  1: Camera tracking only      
  2: Camera tracking and       
     following                 
  3: Camera tracking and       
     following (world frame)   
                               
  T: Toggle smooth tracking    
  F: Toggle smooth following   
                               
  Track offset                 
  u/U: +- 0.1 on X             
  i/I: +- 0.1 on Y             
  o/O: +- 0.1 on Z             
                               
  Follow offset                
  j/J: +- 0.1 on X             
  k/K: +- 0.1 on Y             
  l/L: +- 0.1 on Z             
===============================
```
[image]

Works as expected

csimon-be, 2024-09-09:
Hi @azeey-test,
It works as expected.
@Combuster54, it is best to follow the process and avoid checking already assigned issues to prevent duplicating work. Thank you for your comment.

iche033, 2024-09-09:
thanks for testing

closed 2024-09-09 by iche033