gazebosim/gazebo_test_cases #1476: gz-rendering: Text geometry

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/rendering/9/text_geom.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/19_text_geom_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

j-rivero, 2024-09-12:
Build is broken on Windows:
```
cl : command line  error D8021: invalid numeric argument '/Wno-deprecated-declarations' [C:\Users\josel\code\conda\gz
-ws\src\gz-rendering\examples\text_geom\build\text_geom.vcxproj]
Done Building Project "C:\Users\josel\code\conda\gz-ws\src\gz-rendering\examples\text_geom\build\text_geom.vcxproj" (
default targets) -- FAILED.
```

Easy to fix. But more problems with compilation exists:
```diff
index 7149d1df..292db5de 100644
--- a/examples/text_geom/Main.cc
+++ b/examples/text_geom/Main.cc
@@ -18,7 +18,7 @@
 #if defined(__APPLE__)
   #include <OpenGL/gl.h>
   #include <GLUT/glut.h>
-#elif not defined(_WIN32)
+#else
   #include <GL/glew.h>
   #include <GL/gl.h>
```

At the end the executable works but display a black screen instead of the "Hello World" example.  The `GlutWindow.cc` has some empty conditionals for WIN32 so I'm afraid that this probably not going to work out-of-the-box.

azeey, 2024-09-13:
Can you try including `windows.h` before the gl includes?

closed 2025-08-27 by azeey