gazebosim/gazebo_test_cases #1483: gz-rendering: Render order

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/rendering/9/render_order.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/21_render_order.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

athenaz2, 2024-09-11:
I can do this :)

athenaz2, 2024-09-12:
Instructions work for me for both SDF and `simple_demo`! However, I did have run `simple_demo` with ogre2 render engine as a CLI param (note that default render engine in program is ogre). Seems to me that macOS does not mesh well with ogre? I noticed this when trying other demo program and SDF worlds with ogre/ogre2.

azeey, 2024-09-13:
Thanks for testing! I'm pretty sure only Ogre2 is supported. Mind opening a PR noting that in the tutorial?

iche033, 2024-09-14:
yeah our ogre1 implementation does not quite work on macOS

athenaz2, 2024-09-18:
Okay, added the note in the instructions - https://github.com/gazebosim/gz-rendering/pull/1069

closed 2024-09-18 by ahcorde