gazebosim/gazebo_test_cases #1484: gz-rendering: Render order

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/9/render_order.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/21_render_order.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

jc-sz, 2024-08-30:
Interested in trying this out.

jc-sz, 2024-08-31:
Instructions work as expected, and I got the same image as shown on the tutorial.

I think we can close this issue. (Screenshot attached)
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde