gazebosim/gazebo_test_cases #1493: gz-rendering: Depth camera

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/9/depth_camera.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/23_depth_camera_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-30:
Interested in this tutorial !

iche033, 2024-08-30:
the instructions to install `freeglut3-dev` and `libglew-dev` dependencies should be there? I see it listed in https://github.com/gazebosim/gz-rendering/blob/gz-rendering9/tutorials/02_install.md#source-installation

avanmalleghem, 2024-08-31:
@iche033 yes I saw it in the source installation but if you install it using binary installation, you don't have these prerequisites installed.

avanmalleghem, 2024-09-06:
@iche033, what do we do for this one ? I create a PR or is it ok like this ?

j-rivero, 2024-09-06:
> ## ``
> 
> I first had an issue `No CMAKE_CXX_COMPILER could be found.` that I solved by running `sudo apt-get update && sudo apt-get install build-essential`.

If you could please add it to the tutorial in a PR that would be great.

> @iche033 yes I saw it in the source installation but if you install it using binary installation, you don't have these prerequisites installed.

Good point. These dependencies are only needed for the examples and don't have them included in the binaries. I think that we need to add them in the tutorials. Could you please include this in the PR?

avanmalleghem, 2024-09-06:
Here we go : https://github.com/gazebosim/gz-rendering/pull/1054

ahcorde, 2024-09-09:
PR is merged, thank you for testing

closed 2024-09-09 by ahcorde