gazebosim/gazebo_test_cases #1495: gz-rendering: Bounding Box camera

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/rendering/9/boundingbox_camera.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials/24_boundingbox_camera_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-09-10:
## Mac OS

### Bounding Box Demo
[image]

sreekanthr, 2024-09-13:
Hi trushant,
I accidentally looked at this without seeing your comment.

And noticed that it segfaults, when we exit using ESC.
For exiting using ESC the focus must be on the bounding box window.

Please let me know if you plan to debug it or would be happy for me to do it

sreekanthr, 2024-09-13:
It seems, this is issue is present in all/most examples.
So I suppose it is not related to the tutorial party and can be handled seperately.
Tagging @azeey  for his inputs/decision.

azeey, 2024-09-13:
Thanks for testing!! If the crash on exit happens on most examples, do you mind opening an issue on gz-rendering? We can go ahead and close this issue.

closed 2024-09-13 by azeey

sreekanthr, 2024-09-13:
Created issue [#1056](https://github.com/gazebosim/gz-rendering/issues/1056)