gazebosim/gazebo_test_cases #1499: gz-sensors: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/sensors)
* [Source](https://github.com/gazebosim/gz-sensors/blob/gz-sensors9/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

avanmalleghem, 2024-09-01:
There is a typo in "https://gazebosim.org/libs/sensors/#code-of-contributing" link. It should be "contributing" of course. Will open a PR tomorrow.

Also, I was wondering if it is normal that links are towards version 8 instead of 9 (for installation for example). I guess that's because we are on main, it will be updated when version 9 is released ?

avanmalleghem, 2024-09-02:
PR opened : https://github.com/gazebosim/gz-sensors/pull/468

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde