gazebosim/gazebo_test_cases #1502: gz-sensors: Bounding Box Camera in Gazebo

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sensors/9/boundingbox_camera.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/boundingbox_camera.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-08-29:
Interested

Mechazo11, 2024-09-06:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
    * Works and after adding <path> tag, saved files can be visualized.

closed 2024-09-06 by iche033