gazebosim/gazebo_test_cases #1508: gz-sensors: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sensors/9/installation.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-08-29:
Interested

Mechazo11, 2024-09-06:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
    * Build from source works without any issues in Ubuntu 24.04
    
    * ```sudo apt install libgz-sensors9-dev`` throws an **unable to locate** error.  I checked the cmake from source, it states the library's name is `gz-sensors9`. So I tried to test with `libgz-sensors9-dev` to see if would work or not. 

---

iche033, 2024-09-06:
`libgz-sensors9-dev` has not been released to the stable repo it. It's currently only available in the pre-release repo. So that's expected. Apart from this error, are the other issues with this tutorial?

Mechazo11, 2024-09-06:
@iche033 not to knowledge, but I just saw the last two make test commands that I might need to go back to. I will do those later and report back.