gazebosim/gazebo_test_cases #1509: gz-sensors: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sensors/9/installation.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-11:
Interested in !

avanmalleghem, 2024-09-11:
Blocked by the same kind of issue than on https://github.com/gazebosim/gazebo_test_cases/issues/1533

avanmalleghem, 2024-09-12:
I created a PR : https://github.com/gazebosim/gz-sensors/pull/470

ahcorde, 2024-09-13:
PR is merged, Thank you for testing!

closed 2024-09-13 by ahcorde

avanmalleghem, 2024-09-13:
@ahcorde can you assign it to me please ?