gazebosim/gazebo_test_cases #1510: gz-sensors: Introduction

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/sensors/9/introduction.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/intro.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Yadunund, 2024-09-02:
## Step 1

From the tutorial's landing page, there is a link at the top to continue onto the first tutorial `Next tutorial: Installation`.

[image]

However, the [Installation](https://gazebosim.org/api/sensors/9/installation.html) tutorial page and subsequent ones do not have a links to the next tutorial which could leave users confused on what to try next.

[image]

ahcorde, 2024-09-03:
Thank you for testing!

closed 2024-09-03 by ahcorde